iMike78/nest-mini-drop-in-pcb #32: (Suggestion) Position of usb-c

GegoSK, 2025-09-19:
Probably too late, but is it possible to place the usbc connector at position, that the cable could be connected and pulled out around the speaker through the hanging hole? I tried to do something like that with onju, but microusb connector is to tall. There should be enought space for cable around speaker. It could be useful for firmware upload in case uploading via wifi is not working or possible. 

I sent you a coffee. Thank you for this project!

PS: I read in another topic, where you mentioned "_replacement of different smart speakers_". Just a inspiration - https://www.mi.com/global/product/mi-smart-speaker. I have one, it has better sound than Google nest.

GegoSK, 2025-09-27:
> First of all, there is a reason why the USB-C connector in vertical position, and we have discussed it earlier. Second, I like your suggestion about the Xiaomi speaker, but as I don't own one, I can't work on it.

Just out of curiosity, what was the reason?

Would you need some information or photos firstly? Or should I provide you one speaker? ;-)  I found second hand on sale.

iMike78, 2025-09-27:
The main reason is printed on the PCB as well. The POWER_IN and USB can not be connected in the same time, so the position of the port forcing you to remove the PCB (and disconnect the FPC) before you connect the USB. Your use case scenario is specific.
The other reason is as we need a full version of USB-C connector because of the flashing capability, the horizontal version of that won't fit into the case as that one is way taller than the micro USB.

Before I start to work any other variant of smart speaker I'll need to confirm this one working. So at this time there is no reason to source one.